ros-controls/ros_control #402: Why is `realtime_tools` in a separate repo?

matthew-reynolds, 2019-12-21:
Since #320, `realtime_tools` is not used by any package within the `ros_control` metapackage, but is a dependency of the metapackage itself.

> The dependency in `ros_control` is there since that's a metapackage, `realtime_tools` is part of the `ros_control` bundle, every controller uses it for instance.

_Originally posted by @bmagyar in https://github.com/ros-controls/ros_control/issues/320#issuecomment-381027936_

This is true - every controller depends on it directly, except `position_controllers` (which depends transitively via `forward_command_controller`)  and `rqt_joint_trajectory_controller` (which doesn't really count since it's not a controller itself, just a frontend).

Is there a reason it lives in a separate repo instead of in this repo? I'm not really proposing we move it, since that breaks the git history without much/any benefit, mostly just curious.

---

As an aside, if you're curious like I was, `control_toolbox` is only used by 4 `ros_controllers` packages:
 - `effort_controllers`
 - `gripper_action_controller`
 - `joint_trajectory_controller`
 - `velocity_controllers`

Makes sense why this is not part of the `ros_control` metapackage itself and lives in a separate repo :+1:

matthew-reynolds, 2020-02-06:
Both `control_toolbox` and `realtime_tools` were pulled out of the repo in b8d678f6 (June 2013), but there's no PR or description associated with the commit as far as I can tell.

bmagyar, 2020-02-07:
I think that the idea was that both of these packages serve a general purpose, they have some possibility for use outside of `ros_control` while the packages that were left in the bigger repos are tightly bound to one another.

matthew-reynolds, 2020-02-07:
> These packages serve a general purpose

Gotcha, that makes sense. Thanks! Closing this issue now.

Can you please label this issue as a [**question**](https://github.com/ros-controls/ros_control/issues?q=is%3Aissue+label%3Aquestion) in case someone else has the same question as me down the road? Always good to keep things organized.

closed 2020-02-07 by matthew-reynolds