ros-controls/ros_control #403: Modernize the codebase using C++11/14 features

bmagyar, 2019-12-27:
A non-comprehensive list of TODOs:
* [ ] replace `boost` with `std` whenever possible
* [ ] replace iterator and `BOOST_FOREACH` with range-based `for` whenever possible
* [ ] replace `enum`s with `class enum`s whenever possible
* [ ] use `override` in derived functions
* [ ] watch out if you can add more const!

bmagyar, 2019-12-27:
FYI @matthew-reynolds please feel free to add your ideas on top!

matthew-reynolds, 2019-12-30:
Couple of other thoughts:

 - [ ] use `= default` and `= delete` where appropriate
 - [ ] use `noexcept` where appropriate
 - [ ] use `final` where appropriate
 - [ ] replace `typedef` with `using`
 - [ ] use `auto` when it improves readability (e.g. rather than spelling out the exact iterator when iterating over stl containers, etc)

And some low-priority/more controversial changes:
 - [ ] use pass-by-value and `move` where appropriate (e.g. rather than pass-by-const-ref for some parameters in constructors)
 - [ ] use default member initializers (brace-or-equal-initializer) for member variables to simplify constructors (rather than listing out all the 0s and nullptrs in every constructor)

matthew-reynolds, 2019-12-31:
As mentioned [here](https://github.com/ros-controls/ros_control/pull/395/files#r362134419), another potential cleanup:
 - [ ] creating a `.clang-format` and applying it to the whole codebase
 - [ ] perhaps creating a `.clang-tidy` and applying it the whole codebase

(I suppose it might fall a bit outside the scope of "modernizing" since it's more "cleanup". If preferred, I can open a separate issue to track this).

bmagyar, 2020-01-03:
>   (I suppose it might fall a bit outside the scope of "modernizing" since it's more "cleanup". If preferred, I can open a separate issue to track this).

Indeed I have split this into a separate issue here: https://github.com/ros-controls/ros_control/issues/407

suab321321, 2020-01-15:
@matthew-reynolds I would like to contribute

matthew-reynolds, 2020-01-16:
@suab321321 Thanks, I've got PRs either open or in-progress on my end for almost all of these tasks, except for going through with a fine-toothed comb and adding `const`s and, where it improves readability, `auto`. Would certainly appreciate help in this front. I would also appreciate someone looking into where we can leverage move semantics.

On a related note:

@bmagyar I'm sitting on a PR for function cleanup (adding specifiers `override`, `noexcept`, etc and adding `= default` and `= delete`), but I would like to work out the "use default member initializers (brace-or-equal-initializer) for member variables to simplify constructors" task first since it allows several constructors to be eliminated or changed to `= default`. Do you have a preference on brace vs equal initializers for member variables? Braces have the advantage of not allowing narrowing, and although it had some issues in C++11 (see initializer lists), I think those are all resolved in C++14.  I admit, though, I am not an expert here.

1) `int a {0};`
2) `int a = {0};`
3) `int a = 0;`

suab321321, 2020-01-17:
@matthew-reynolds okay

suab321321, 2020-01-19:
@matthew-reynolds I m only making iterators auto will that be sufficient?

matthew-reynolds, 2020-01-23:
> I'm only making iterators auto will that be sufficient?

#413 updated (almost) all for loops to range-based for loops using `const auto&` and `auto&&`, so I believe we are pretty well covered for iterators. At least, covered for iterators in for loops. I should have mentioned this in my previous message, my apologies.

Other than for loops, I haven't looked for places where `auto` would improve readability. We may need guidance from a maintainer on the scope of where they would like `auto`s, but I doubt we will have any hard rules. It is probably mostly a case-by-case judgement, but here are some ideas of places where readability could probably be improved by using `auto`:

 - As the return value of STL functions that return iterators, such as `std::find`, `std::begin`, etc
 - For const references used for convenience/to shorten the name [See Ex 1]
 - For long and/or templated types where the type is obvious

Ex 1:
`const std::vector<InterfaceResources>& c_res = info_it.claimed_resources;` ->
`const auto& c_res = info_it.claimed_resources;`

suab321321, 2020-01-23:
@matthew-reynolds I have a created PR is that okay?

matthew-reynolds, 2020-01-27:
@bmagyar Can you advise on brace vs equal initializers? Looking for your opinion on which of the following you prefer to initialize class members:

1) `int a {0};`
2) `int a = {0};`
3) `int a = 0;`

bmagyar, 2020-01-27:
Personally I'd opt for the simplest option, number 3.

bmagyar, 2020-11-17:
I'm closing this one now, please reopen if there's anything you'd add

closed 2020-11-17 by bmagyar