ros-controls/ros_control #408: Investigate rosunit dependency requirement

matthew-reynolds, 2020-01-12:
Following up from https://github.com/ros-controls/ros_control/pull/404#discussion_r363081046:

When using `catkin_add_XXXtest()` in `CMakeLists.txt`, it's unclear whether we should   `<test_depend>rosunit</test_depend>`. Most documentation seems to say that we should, but the documentation seems old.  In face, the word "rosunit" [doesn't appear in any code in the Catkin repo](https://github.com/ros/catkin/search?q=rosunit&type=Code).

`gmock`, `gtest`, `python_nose`, `python_mock`, etc are all dependencies of `catkin`

The question is whether we need to depend on rosunit at all when using the `catkin_add_XXXtest()` commands. `catkin_lint` seems to think we do, but the `catkin_add_gtest` rule also hasn't been updated in 5 years.

---

See https://github.com/ros-controls/ros_control/pull/404#discussion_r364035240 for a table of packages in the [ROS GitHub organization](https://github.com/ros) which use `catkin_add_gtest()`, and whether they `<test_depend>rosunit</test_depend>` or not.

matthew-reynolds, 2020-01-12:
Out of curiosity, I removed:
 - `/opt/ros/melodic/share/rosunit`
 - `/opt/ros/melodic/lib/python2.7/dist-packages/rosunit`
 - `/opt/ros/melodic/bin/rosunit`

and re-ran `catkin run_tests ros_control`. All the `rostest` tests failed (as expected), but it appears all the `catkin_add_XXXtest`tests succeeded even with `rosunit` missing.

This further leads me to believe that the documentation and `catkin_lint` are just out of date.

matthew-reynolds, 2020-01-13:
> `catkin` itself doesn't need / use `rosunit`. The CMake functions containing `rostest` in the name are provided by the `rostest` package and that package depends on `rosunit`. But if a package depends on `rostest` it doesn't need to declare a dependency on `rosunit` (except if it explicitly uses it somewhere directly).

_Originally posted by @dirk-thomas in https://github.com/fkie/catkin_lint/pull/76#issuecomment-573808752_

Sounds like we are indeed correct, and that the documentation is simply out of date. fkie/catkin_lint#76 is already open to update `catkin_lint`, I will open PRs to update the catkin documentation.

Thanks @bmagyar for digging into this further and @dirk-thomas for explaining the situation.

closed 2020-01-13 by matthew-reynolds

bmagyar, 2020-01-14:
Thanks for following up!

matthew-reynolds, 2020-01-14:
See documentation update in ros/catkin#1046