ros-controls/ros_control #409: Remove _tests packages from metapackage

matthew-reynolds, 2020-01-12:
Following up from https://github.com/ros-controls/ros_control/pull/404#discussion_r363080746:

The `combined_robot_hw_tests` and `controller_manager_tests` packages should not be released and should not be dependencies of the metapackage. If they offered a framework for testing plugins/3rd party code, they would have a reason to be released, but as implemented they are essentially just more tests for their respective packages.

Steps to take:
 - [ ] Remove these packages from the `ros_control` metapackage, certainly in Noetic but ideally also in Kinetic and Melodic
 - [ ] In Noetic, do not release these packages

mateus-amarante, 2020-04-16:
Shouldn't those *_tests packages be test dependencies of the meta package? Like:

```xml
<test_depend>controller_manager_tests</test_depend>	
<test_depend>combined_robot_hw_tests</test_depend>
```

matthew-reynolds, 2020-04-16:
`rosdep` doesn't distinguish between any types of dependencies, it just globs everything together (See [here](https://github.com/ros-infrastructure/rosdep/blob/d1784da592310c7ad8e649530200cdf957a87afc/src/rosdep2/rospkg_loader.py#L143)). `rosinstall_generator` does the same (See [here](https://github.com/ros-infrastructure/rosinstall_generator/blob/b175367a4d39473c77871c91719a39e277d26b64/src/rosinstall_generator/distro.py#L89)).

Since there's no ability to filter out test dependencies, some potentially very heavy dependencies (ie gazebo) will be pulled in. Excluding test depends from the metapackage (And ultimately re-structuring some tests into dedicated packages which can also be excluded) makes it much more convenient for non-desktop installs.

matthew-reynolds, 2020-04-16:
Also, I forget who but someone brought up that it's still worthwhile to release the tests in case anyone wants to build their own tests on top of them, even if they're not part of the metapackage proper. So I'm going to remove the second item from this TODO and close this issue, as it is complete. Heads up @bmagyar

closed 2020-04-16 by matthew-reynolds