ros-controls/ros_control #414: Investigate pluginlib dependency requirements

matthew-reynolds, 2020-01-14:
Follow up from https://github.com/_render_node/MDIzOlB1bGxSZXF1ZXN0UmV2aWV3VGhyZWFkMjI1ODA5NTU0OnYy/pull_request_review_threads/discussion:

Do we need to `<exec_depend>` on header-only packages containing plugin interfaces (ie. `hardware_interface`), or are `<build_depend>` and `<build_export_depend>` sufficient?

I don't know why the `exec_` would be required, but [the pluginlib wiki](http://wiki.ros.org/pluginlib#pluginlib.2BAC8-pluginlib_groovy.Registering_Plugin_with_ROS_Package_System) states that "the [plugin-]providing package must depend directly on the package containing the plugin interface", with the code example containing both `run_` and `build_` dependencies.

I don't know whether the wiki's advice applies to header-only packages or not, I suspect this is unnecessary.

ipa-mdl, 2020-01-14:
If I am not mistaken the `run_depend`/`exec:_depend` is needed for the plugin resolution mechanism.

matthew-reynolds, 2020-01-14:
Good point, I hadn't considered `rospack`'s internals. Thanks for pointing that out!

---

I dug in a little, dropping this here just for completeness.

```
rospack plugins --attrib=plugin controller_interface
```

fails if `controller_interface` is not present. From a quick glance at the rospack source ([rospack.cpp, line 1085](https://github.com/ros/rospack/blob/8c5a699461f32f620204ccfaf93c5e5181f20f14/src/rospack.cpp#L1085)), it looks like this is because `rospack` resolves plugins by:
1) Locating the plugin interface package (e.g. `controller_interface`)
2) Listing all packages that depend on this plugin interface package (e.g. the controllers in `ros_controllers`)
3) Scanning the `package.xml` of each of these "depends-on" packages

If the dep was only `build_` and `build_export_depend`, step 1) may fail.

I don't know how `pluginlib`, `class_loader`, and `rospack` all play together, but if `rospack` fails like this, it seems likely the others would fail as well. Even if the other tools don't fail, we still don't want to break `rospack`, so the `exec_depend` is indeed required.

closed 2020-01-14 by matthew-reynolds