ros-controls/ros_control #420: Preliminary Noetic Roadmap

matthew-reynolds, 2020-01-27:
Just want to start a place to collect changes for Noetic ahead of its release, since there doesn't seem to be a tracking issue yet. Happy to close in favour of a more formal roadmap if a maintainer makes one.

 - Do not release `_tests` packages (Related: #409)
 - Replace enums with enum classes (#412)
 - Remove non-const `ForceTorqueSensorHandle` constructor previously left in for ABI compatibility. Probably should have been removed in Melodic, but it got missed (Related: #283)

bmagyar, 2020-02-09:
* Review all Python code and ensure Python3 compatibility

sloretz, 2020-03-11:
* Bump CMake version to avoid CMP0048 warning on ROS build farm #427

sloretz, 2020-04-13:
Friendly ping, how's progress on a ROS Noetic release? It looks like all of `ros_control`'s dependencies are released, and [it's currently blocking 110 repos](http://repositories.ros.org/status_page/blocked_releases_noetic.html?q=id%3D%22ros_control%22&s=6&r=1).

matthew-reynolds, 2020-04-13:
The main blocker is #429, which is waiting on the next Catkin release (https://github.com/ros/rosdistro/pull/24386)

bmagyar, 2020-04-15:
CI is coming alive:
https://travis-ci.org/github/ros-controls/ros_control/jobs/675379090

bmagyar, 2020-04-22:
https://github.com/ros-controls/ros_control/pull/439

bmagyar, 2020-04-23:
A follow-up ticket for after the release: https://github.com/ros-controls/ros_control/issues/448

bmagyar, 2020-04-23:
@sloretz  you can find the initial release here: https://github.com/ros/rosdistro/pull/24541
@matthew-reynolds I'll create a new issue for the next release, a few non-breaking PR-s we should add to noetic soonish

bmagyar, 2020-04-23:
https://github.com/ros-controls/ros_control/issues/449

closed 2020-04-23 by bmagyar