ros-controls/ros_control #424: Transport error and gazebo segmentation fault.

johuber, 2020-02-07:
I am getting an exception  at [this line](https://github.com/ros-controls/ros_control/blob/10ff185bc31113ca434e3169e0e3b66193f4eef8/controller_manager/scripts/spawner#L199), since upgrade my linux packages today. (using ros-melodic).

Immediately after a segmentation fault occurs. Is anyone else facing similar issues?

matthew-reynolds, 2020-02-07:
Looks like this it may be the same problem mentioned in #418

johuber, 2020-02-07:
So the full error message is this:
```
[INFO] [1581065620.621149][/controller_spawner][/opt/ros/melodic/lib/controller_manager/spawner:184]: Loading controller: joint_state_controller
[INFO] [1581065620.647413][/controller_spawner][/opt/ros/melodic/lib/controller_manager/spawner:192]: Controller Spawner: Loaded controllers: blitz_one_tricycle_drive_controller, joint_state_controller
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 207, in <module>
    if __name__ == '__main__': main()
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 199, in main
    resp = switch_controller(loaded, [], 2, False, 0.0)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 439, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 529, in call
    raise ServiceException("transport error completing service call: %s"%(str(e)))
rospy.service.ServiceException: transport error completing service call: unable to receive data from sender, check sender's logs for details
[INFO] [1581065676.594091][/controller_spawner][/opt/ros/melodic/lib/controller_manager/spawner:55]: Shutting down spawner. Stopping and unloading controllers...
[INFO] [1581065676.595180][/controller_spawner][/opt/ros/melodic/lib/controller_manager/spawner:64]: Stopping all controllers...
[WARN] [1581065676.611304][/controller_spawner][/opt/ros/melodic/lib/controller_manager/spawner:72]: Controller Spawner error while taking down controllers: unable to connect to service: [Errno 104] Connection reset by peer
Segmentation fault (core dumped)
```

Note that about 30s pass, before the exception is thrown.
I will have to look deeper into it first myself, I realized we also may have some non-updated versions.

I will get back to you if I keep having the issue.

johuber, 2020-02-07:
Apologize, it was an internal issue. I did not realize we had some outdated forks which lead to the issue. Interesting they only occurred now.

Thanks anyway!

closed 2020-02-07 by johuber