ros-controls/ros_control #442: Controller Manager CLI Broken Melodic

danem, 2020-04-16:
rosrun controller_manager controller_manager list

just hangs indefinitely. 

    rosservice call /robot/controller_manager/list_controllers

works however...

bmagyar, 2020-04-17:
Please provide sufficient details whenever you are reporting an error.
Was it installed or built from source?
Which version of the package or the repo?

danem, 2020-04-17:
Sorry for not being more thorough. Here is my environment:

I installed the package via apt with `sudo apt-get install ros-melodic-controller-manager`. Dpkg and the `package.xml` file tells me that I have 0.16 installed currently.

I am currently running ros melodic with most everything installed via apt.

bmagyar, 2020-04-19:
We have a test for the CLI and it passes for Melodic. Can it be that you are running this on Python3 perhaps?
I am right now fixing bytes vs strings for Noetic on that but it shouldn't affect Melodic.
Try running `controller_manager_scripts` from the `controller_manager_tests` package to verify this.

danem, 2020-04-27:
Sorry for the late response.

I'm running python 2. And sorry, but I was unable to find the `controller_manager_scripts` you refer to. I've installed `ros-melodic-controller-manager-tests`.