ros-controls/ros_control #443: Controller Manager Freezing on multiple controller switches

krishnachaitanya9, 2020-04-17:
My PC Specifications:
OS: Pop!_OS 18.04 LTS x86_64 
Host: Razer Blade 3.04 
CPU: Intel i7-9750H (12) @ 4.500GHz
GPU: NVIDIA NVIDIA Corporation Device RTX 2070
RAM: 16GB
ROS Version: Melodic

I have a gazebo simulation of Franka Emika robot, and when I switch the controllers multiple times it freezes. When I turn on the DEBUG statements It prints debug statement at this line: https://github.com/ros-controls/ros_control/blob/melodic-devel/controller_manager/src/controller_manager.cpp#L610 and doesn't move forward from there. I guess it's getting stuck in the while loop in line 611. I don't know what might be causing this problem. I don't know what extra details to post, but I will post whatever you guys ask for. Thank you!

bmagyar, 2020-04-19:
Is this sim publicly available? If yes, could you please post some steps to reproduce the issue?

krishnachaitanya9, 2020-04-21:
Hi @bmagyar. Below are the steps to reproduce my problem using the simulation I use.
```bash
mkdir freezing_issue && cd freezing_issue
mkdir src && cd src
git clone --single-branch --branch AddFreezingScript https://github.com/HIRO-group/ros_robotic_skin.git
cd ros_robotic_skin
./install.sh
```
The freezing issue would be the catkin workspace name. Once you source /PATH/TO/freezing_issue/devel/setup.bash by adding it to your bashrc (This step is basically activating the catkin workspace devel), you can run the simulation using:
```bash
roslaunch ros_robotic_skin simulation.launch
```
Now a gazebo window should open up with the simulation of franka panda robot. Now to reproduce the actual freezing you need to run:
```bash
rosrun ros_robotic_skin infinite_switches.py
```
We have created some controllers like Position Controllers and Velocity Controllers, and we need to repeatedly switch in between them. The `infinite_switches.py` accurately simulates the problem I am going through. 
Kindly let me know if I face some issues reproducing the problem, and I will try to answer right away. Thank you so much for your help.

krishnachaitanya9, 2020-05-19:
Hi, @bmagyar kindly lemme know if I can be of any help in reproducing the issue. I will attend to it right away. Thanks!