ros-controls/ros_control #445: [rqt_controller_manager] Cannot handle controller after loading it

RobertWilbrandt, 2020-04-22:
rqt_controller_manager doesn't seem to handle controller loading correctly. After Loading a controller, the **state** column correctly switches to *initialized*, but the marker doesn't switch from grey to red and it is not possible to right-click on the controller to perform further switches.

[image]

When the controller gets switched directly to a different state using a service the gui updates correctly, but unloading and loading the controller again brings back the problem. *Load and Start* also works correctly.

I created a small reproduction repo at [rqt_controller_bug](https://github.com/RobertWilbrandt/rqt_controller_bug), but i also encountered this with different robot arms (franka panda, ur5) and some custom nodes. I tried to build all relevant projects directly from git in *melodic_devel*, you can find a wstool file in the linked repo. I think the problem appeared at the end of December 2019, but that is quite uncertain.

I am currently running on ROS melodic on Ubuntu 4.15.0-96-generic, but as this problem seems to appear for colleagues on different systems too that should not really matter too much.

I'll look into solving this and creating a PR by myself, but don't know when i will be able to do so.

bmagyar, 2020-04-22:
Potential culprot https://github.com/ros-controls/ros_control/pull/391/files ?

RobertWilbrandt, 2020-04-22:
Thank you for the pointer, that indeed seems like the source for this behavior. The PR changed the possible strings in [the controller state](https://github.com/ros-controls/ros_control/blob/ee00d63ed4e063fafb5e75672511597d14df1e4c/controller_manager_msgs/msg/ControllerState.msg#L2). 32c74f6 then added the same grey icon as for *uninitialized* to the new *initialized* state, but didn't add the appropriate context menu. The PR added more states which simply don't have an item and context menu right now (probably understandable, as all of them shouldn't appear for long).

I will create a PR that adds the context menus and changes icons for all states. The only open point would be whether to change *initialized* back to red or to change colors slightly so you can differentiate between stopped and just loaded controllers (e.g. orange for *stopped* and red for *loaded*, might be useful but i'm not sure if i like that myself).

PR #450 (2020-04-25): Fix rqt displaying and handling of 'initialized' controllers
Fixes #445. This restores the rqt gui behavior before #391, which changed possible ControllerState states. I don't really understand enough about the other states (unknown, waiting, aborted) to know if any action for them is useful.
closed 2020-04-25 by bmagyar