ros-controls/ros_control #451: Spawning stopped controllers sets them to state `initialized` instead of `stopped`

Martin-Oehler, 2020-05-08:
Hello everyone,

I tried to spawn controllers stopped with
```
rosrun controller_manager spawner --stopped flipper_front_velocity_controller
```
However, this puts the controller from state `uninitialized` to `initialized` instead of the expected state `stopped`. Did this behavior change somehow?

This leads to the problem, that you are no longer able to start a controller using the `rqt_controller_manager` plugin, which does not offer a right-click menu for items in the `initialized` state.

Is this the expected behavior? Thanks for your help.

Best regards,
Martin

RobertWilbrandt, 2020-05-08:
The behavior was changed in [this PR](https://github.com/ros-controls/ros_control/pull/391), which added additional states. I also found the rqt problem (see #445), it should be fixed since #450. Could you try using the current ```rqt_controller_manager``` version (i.e. from source) and check if this works for you?

Martin-Oehler, 2020-05-08:
Yes, this did indeed work. Thank you for the fast response.

closed 2020-05-08 by Martin-Oehler

bmagyar, 2020-05-09:
Thanks guys for both reporting & providing the fix. I've made the release for Melodic: https://github.com/ros/rosdistro/pull/24803

Should get around with a new release with Noetic containing this fix too today