ros-controls/ros_control #463: Register to position and effort handle

arsh09, 2020-06-11:
Hello, 

I am sorry if this is repetitive. 

I wrote a hardware interface class for a robotic arm with Joint Position Interface and Joint state interface. I have written its RobotHW class and exposed read , write method. It works perfectly. (Thank you) 

Now, I would like to switch controllers on run time from Position to Effort. Arm joints (motors controller) can be switched really easily (using the same RobotHW class, from low level position control to current control) though I was wondering what is the method to approach such a task. In saying that, I have two questions:

1  - In my understanding, I have to register the arm resources to Joint Effort Interface (similar to how I did for Joint Position and State Interface). But I do not get it how to register the same resources to two different interfaces (in the same class-method).

2 - How would I know when the user switches the controller from controller manager service? Is there any class method from control manager ? 

I look forward to some suggestions. 

Thank you

mateus-amarante, 2020-08-02:
> 1 - In my understanding, I have to register the arm resources to Joint Effort Interface (similar to how I did for Joint Position and State Interface). But I do not get it how to register the same resources to two different interfaces (in the same class-method).

You can register multiple handles for the same resource. Example:

```cpp
hardware_interface::JointStateInterface jnt_state_interface;
hardware_interface::PositionJointInterface jnt_pos_interface;
hardware_interface::EffortJointInterface jnt_eff_interface;

double pos, vel, eff, pos_cmd, eff_cmd;

hardware_interface::JointStateHandle state_handle("joint", &pos, &vel, &eff);
hardware_interface::JointStateHandle pos_cmd_handle(state_handle, &pos_cmd);
hardware_interface::JointStateHandle eff_cmd_handle(state_handle, &eff_cmd);

jnt_state_interface.registerHandle(state_handle);
jnt_pos_interface.registerHandle(pos_cmd_handle);
jnt_eff_interface.registerHandle(eff_cmd_handle);
```

> 2 - How would I know when the user switches the controller from controller manager service? Is there any class method from control manager ?

The controller manager provides a [method](https://github.com/ros-controls/ros_control/blob/ab33807ad73c1aaa66410d2282db0aa99b423c82/controller_manager/include/controller_manager/controller_manager.h#L129-L146) and a [ROS service](http://wiki.ros.org/controller_manager?distro=noetic) to switch controllers that you can call on runtime. If you need to do something at low level to switch the controllers, just implement the [switch interface ](https://github.com/ros-controls/ros_control/blob/ab33807ad73c1aaa66410d2282db0aa99b423c82/hardware_interface/include/hardware_interface/robot_hw.h#L120-L136) of RobotHW class.

I suggest watching [this ROSCon Talk](https://vimeo.com/107507546) and reading the source code of ControllerManager and RobotHW classes to learn more about the ros_control structure.

I hope I could help you.

arsh09, 2020-08-02:
@mateus-amarante 

Thank you for your response. Yes, I recently implemented this and it works really well. I implemented the doSwitch function for switching as per your suggestion. 

I was also wondering if I have an arm and a gripper, and they both use same actuators (with a CAN bus). Can I use the same ros control C++ file to register the handles. If I spawn 2 position controllers, how will I know which joints to register. or Do I need to implement the combined HW interface?

closed 2020-08-02 by arsh09