ros-controls/ros_control #464: python hardware interface binding

a-z-e-r-i-l-a, 2020-06-13:
I want to make a python interface binding to a robot hardware interaface that we have, and the choices seems to be using either Cython, boost/python, swig, and etc.

Do you have a recommendation which one of these might be better and have a lower latency? my goal is to invoke controller update, from python while sending robot state also from python along with getting control commands from the c++ hardware interface.

Thanks.

bmagyar, 2020-06-14:
That sounds fairly interesting! My go-to would be Pybind11, of course depending on the Ubuntu/ROS version this may not be available.

a-z-e-r-i-l-a, 2020-06-14:
I use ROS Melodic and Ubuntu 18.04. How about Cython or Boost/python? I thought maybe these might have a better latency. Do you think it might be time consuming for a fairly beginner in C++ to make prepare this binding?