ros-controls/ros_control #465: UR10e connected the pc,but doesn't move it

tt2cloud, 2020-06-15:
> Hi, Dear all, I found a problem,the UR10e can connect the PC,I can see the gesture of ur10e under rviz, but when I use moveit to move it, I find that the real Hardware cannot move, and the error is as follows：

> terminal 1:roslaunch ur_robot_driver ur10e_bringup.launch robot_ip:=**.**.**.**

SUMMARY
========

PARAMETERS
 * /controller_stopper/consistent_controllers: ['joint_state_con...
 * /force_torque_sensor_controller/publish_rate: 500
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /hardware_control_loop/loop_hz: 500
 * /hardware_interface/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/type: velocity_controll...
 * /joint_state_controller/publish_rate: 500
 * /joint_state_controller/type: joint_state_contr...
 * /pos_traj_controller/action_monitor_rate: 10
 * /pos_traj_controller/constraints/elbow_joint/goal: 0.1
 * /pos_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/goal_time: 0.6
 * /pos_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /pos_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /pos_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /pos_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /pos_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /pos_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /pos_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /pos_traj_controller/joints: ['shoulder_pan_jo...
 * /pos_traj_controller/state_publish_rate: 500
 * /pos_traj_controller/stop_trajectory_duration: 0.5
 * /pos_traj_controller/type: position_controll...
 * /robot_description: <?xml version="1....
 * /robot_status_controller/handle_name: industrial_robot_...
 * /robot_status_controller/publish_rate: 10
 * /robot_status_controller/type: industrial_robot_...
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /scaled_pos_traj_controller/action_monitor_rate: 10
 * /scaled_pos_traj_controller/constraints/elbow_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/constraints/goal_time: 0.6
 * /scaled_pos_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /scaled_pos_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/joints: ['shoulder_pan_jo...
 * /scaled_pos_traj_controller/state_publish_rate: 500
 * /scaled_pos_traj_controller/stop_trajectory_duration: 0.5
 * /scaled_pos_traj_controller/type: position_controll...
 * /scaled_vel_traj_controller/action_monitor_rate: 10
 * /scaled_vel_traj_controller/constraints/elbow_joint/goal: 0.1
 * /scaled_vel_traj_controller/constraints/elbow_joint/trajectory: 0.1
 * /scaled_vel_traj_controller/constraints/goal_time: 0.6
 * /scaled_vel_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /scaled_vel_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.1
 * /scaled_vel_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /scaled_vel_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.1
 * /scaled_vel_traj_controller/constraints/stopped_velocity_tolerance: 0.05
[... 370 lines cut ...]
[ INFO] [1592186899.903129697]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186901.386982485]: Replanning: yes
[ INFO] [1592186902.782810630]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186903.865067708]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186905.108830083]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186907.609091666]: Looking around: yes
[ INFO] [1592186914.485051143]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186915.356293093]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186915.577693663]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186917.639029757]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186929.123774766]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186930.265397395]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186931.238410735]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186939.893111610]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186941.204564882]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186968.658662198]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186969.758597597]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186970.864241100]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186972.708550601]: ABORTED: Solution found but controller failed during execution
[image]

> terminal 4:rosrun rqt_joint_trajectory_controller rqt_joint_trajectory_controller

[image]
Could you help me solved the pronlems, Thank you advance!!@gavanderhoorn@fmauch

tt2cloud, 2020-06-17:
Brother, I tested the script file `rosrun ur_driver test_move.py` today,but I found the real Hardware can't move.
[image]
@gavanderhoorn @fmauch
Could you help me, thanks in advance!

bmagyar, 2020-06-17:
Sorry but I don't really know why you are posting this issue on `ros_control`. As far as I can tell you have an issue with the UR drivers.

tt2cloud, 2020-06-17:
> Sorry but I don't really know why you are posting this issue on `ros_control`. As far as I can tell you have an issue with the UR drivers.

tt2cloud, 2020-06-17:
> Sorry but I don't really know why you are posting this issue on `ros_control`. As far as I can tell you have an issue with the UR drivers.

yeah , I issued this question to [(https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/200)]
thank you, Brother