ros-controls/ros_control #469: Effort controller with gravity compensation

alexswerner, 2020-07-21:
Hello,

I'm running a bunch of different Gazebo simulation with robot arms. To have working contact dynamics in gazebo I'm using the joint effort interface to define the controller. However this gives me a PID control for each joint. Now I want relatively precise tracking of the joint position so i increase the gains to e.g. 10^8 but still get errors from the joint trajectory controller that the robot is not tracking the desired trajectory. Of course that happens because the controller has neither gravity compensation or feed forward terms (M*ddq). What would be your approach to fix that? Are there any effort to create a decent arm controller? Or should i start one?

Great work so far guys!

mateus-amarante, 2020-07-25:
Hello, let me try to help you.

The gazebo_ros_control's default RobotHW does not apply PID control for the effort hardware interface, it simply applies the target effort to the simulated joint. The PID params are valid for position and velocity interfaces. Internally, it computes the position/velocity tracking errors and applies PID control to define effort values for the joints.  Check [these](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/kinetic-devel/gazebo_ros_control/src/default_robot_hw_sim.cpp#L216-L232) [parts](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/kinetic-devel/gazebo_ros_control/src/default_robot_hw_sim.cpp#L313-L385) of gazebo_ros_control code.

As far as I know, gravity compensation is a centralized control approach that requires the knowledge of all the joint positions to define the G(q) matrix. So, you need a "multi-joint scoped" solution. In this way,  I suggest creating a multi-joint custom controller for the effort hardware interface that should compute the effort commands for all the joints with all the terms you want (PID + Gravity Compensation + Feedforward + ...).

There is also a gravity_compensation [ROS package](http://wiki.ros.org/gravity_compensation) and a [Gazebo plugin](http://gazebosim.org/tutorials?tut=gravity_compensation&cat=plugins) but I'm not sure if they might be useful (I have just googled it).

alexswerner, 2020-08-03:
Many thanks for the feedback, the gazebo plugin seems indeed to be a way to implement this. I also found https://github.com/pal-robotics/gravity_compensation_controller_tutorial which seems to implement this as well. But just to reiterate the problem from my perspective - please let me know where I am going wrong

1. Fixed base manipulator robot with gripper in gazebo.
2. When the arm is in PositionJointInterface mode, stable grasping is impossible, the objects always slip -> need EffortJointInterface
3. Running standard ros_control JointTrajectoryController with integrated PID gives significant difference between desired and actual joint position due to lack of gravity compensation.

Is this a correct description of the situation?

mateus-amarante, 2020-08-04:
Good find. It seems this gravity_compensation_controller_tutorial does exactly what I suggested. It seems a great start. Thanks for sharing it.

Talking about your situation's description:

I do not have practical experience with grippers, but looking at [gripper_action_controller](https://github.com/ros-controls/ros_controllers/tree/noetic-devel/gripper_action_controller), it should work with both interfaces (better with effort interface to consider max_effort input). The performance will depend on tuning and the target commands.

I think the lack of gravity compensation is not the reason for the bad performance you are facing. I think your control constants might not be adequate for your model. I suggest reviewing your URDF model, looking for unrealistic params (like inertia and effort limits), and strugling a bit more with tuning. [This package](url) provides nice tools to tweak pid params.