ros-controls/ros_control #480: Position controlller sends mismatching command on load

arsh09, 2020-10-21:
Hello, 

I wrote a hardware interface class for my robotic arm. It uses Position Controller on initialisation and later on, it can be switched to Effort and Velocity controllers (I used preparedSwitch and doSwitch functions). 

During resource registration (in RobotHW construction function), I set the command variabled (cmd) equals to the position state variable (pos). But when the controller starts (using launch file) the robotic arm moves a little.

During debugging, I found that after the position controller is completely loaded, the cmd variable changes for some reason and it causes the RobotHW write class to write a new position to each joint. 

The picture attach shows that right after starting the joint state and position controllers, the CMD variable changed a little (currently I am debugging and moving only one joint). 

This is a little confusing as I have tried a few things and nothing seems to work: 

1 - I changed the cmd variables for each type of controller while registering the resources in RobotHW constructor 
2 - Instead of making cmd[i] = pos[i], I made it equal to 0 (so that controller always start the arm in home position) and that did not work also

I noticed that 'Start controller' ros log is from controller manager / scripts / spawner.py which calls the a service in controller_manager/src/controller_manager/controller_manager.py. But there is no variable that affects the cmd variable.

[image]

Ros terminal logs

[Uploading logs.txt…]()

Thank you

bmagyar, 2020-10-21:
The spawner script will load and start all controllers, meaning that the position controller will start writing something to your underlying commands structure.

Also, please add your ROS distribution and the version of the ros_control and all ros_controllers packages you are using in future bug reports, wherever you do it. That way we don't have to guess.

arsh09, 2020-10-21:
@bmagyar 

Thank you for your response. I added the distribution information. 

I realise that position controller (which is spawned in non-stop mode as oppose to velocity and effort controllers) starts writing to my hardware (i.e. continuously calling read -> update -> write functions. What I do not understand is why and where it (or something else) changes the cmd variable?

As I mentioned, I set the command variable to be equal to position I received. I checked all the conversions for gearing and Revolution to radians (for getting position state) and opposite for radian to revolutions for writing (i.e. setting position on hardware). There is no place left where I am setting the value of command variable except in constructor.

arsh09, 2020-10-22:
I solved the problem by changing the configuration file for Position Controller.

I removed all parameters from configuration file except goal time (set to 0.5 seconds) and stopped_velocity_tolerance (set to 0.0) 

Thank you

closed 2020-10-22 by arsh09